hatem-darweesh/assuremappingtools #1: Convert Vector Map to Opendrive

monkeykane, 2019-07-24:
Hi @hatem-darweesh 
Thank you for offering the covert tool. 
So far I am working on about how to convert Vector Map to Opendrive.  But I found some important information are missing, such as center line, relation of 'road' and 'lanes'. 
I'd appreciate it if there are any tips or advice to do it? 
Thank you !
BR,
wkwu

P.S when I use assure map editor import some Vector Map, it is always crashed. Some info in Terminal as follow:
----
Before The OpenGL Main Loop
 >> Loading vector map data files ... 
 >> Extracting Lanes ... 
Segmentation fault (core dumped)
----

hatem-darweesh, 2019-08-07:
@monkeykane can you share the data with me to test this problem.

monkeykane, 2019-08-07:
@hatem-darweesh 
https://github.com/lgsvl/autoware-data/tree/master/data/map/vector_map_sim_sf_portion

hatem-darweesh, 2019-08-08:
@monkeykane , something wrong with these maps ! 
I fixed the segmentation fault, but there is some king of shift in the points.csv data file. points ID starts from 2, if changed to start from 0 , the map will visualize correctly. 
also I tried to load using autoware, but autoware failed to load the map.

monkeykane, 2019-08-09:
@hatem-darweesh 
Thanks for your feedback. I used lgsvl-autoware to load this map, it worked, but frame rate in rviz is very slow, only 1-2 fps

hatem-darweesh, 2019-08-13:
@monkeykane this map contains too many waypoints, it will always be slow in visualization. 
for the reading part in the map editor , I added another menu item to use different parser.

closed 2019-08-13 by hatem-darweesh