hatem-darweesh/assuremappingtools #8: Binaries and Dependencies

mjshiggins, 2020-04-29:
I was curious about why this project depends on ROS1 (specifically when the software it's looking to augment, Autoware, targets ROS2), but when I wanted to dive in to investigate I realized that this entire project is a conglomeration of compiled libs and binaries. While I appreciate the effort you've put into this project, it's impossible for anyone to help out or contribute without access to the source. I've been looking for a tool like this for awhile, and while I've gotten the opportunity to use and develop private implementations at a few different companies, I haven't seen anything open source. It would be great if I could help contribute!

hatem-darweesh, 2020-05-05:
@mjshiggins thank you for you comment, 
To be honest with you, I didn't want to be dependent on ROS at all but: 
- This is project depends on Autoware, because it was built to support mapping for robotics projects that uses Autoware.AI , and Autoware.AI is based on ROS 1. 
- Second essential reason for depending on ROS 1 is using the rosbag player. 

The main target is to automate map creation. so this just first step. 
The mapping project aim is commercial, that is why it is not open source. and that is why it is rare to find an open source mapper. (mapping is a huge business in robotics and autonomous driving)

Thank you for your proposal, you can help me by bug report and suggest new functionality.

mjshiggins, 2020-05-08:
@hatem-darweesh Sorry, if it's going to be closed source than I can't help contribute. I understand the value of mapping tools and am looking to help build something with an open standard. Good luck!

closed 2020-06-28 by hatem-darweesh