hatem-darweesh/assuremappingtools #9: Dependencies aren't fully outlined

mjshiggins, 2020-04-30:
I'm attempting to containerize this application in Docker, but the dependency list seems to be incomplete. I'd like to investigate further but am unable to due to Issue #8 and lack of code. Happy to help be a contributor here to get this running smoothly in Docker.

hatem-darweesh, 2020-05-05:
@mjshiggins WOW , that is great, thanks a lot, 
btw I added native [ubuntu 18.04 build ](https://github.com/hatem-darweesh/assuremappingtools/tree/ubuntu18.04_build). 

what this project needs is a proper linux installer container.

mjshiggins, 2020-05-08:
Closing due to closed source: https://github.com/hatem-darweesh/assuremappingtools/issues/8#issuecomment-625783723

closed 2020-05-08 by mjshiggins