hatem-darweesh/assuremappingtools #12: Seg fault when Loading an .osm that was just created using the tool

popeyedpanther, 2020-06-05:
When I load a .osm file, that I just created using the mapping tool I get a segmentation fault and the program terminates. Here is a link to the osm file: https://www.dropbox.com/s/o3otueh1um6qon0/Test_Map_2.osm?dl=0

The map is very simple. Two lane segments that are linked with the to/from tag. Any help would be appreciated.

robinchen2019, 2020-06-22:
i have the same problem. the .osm is created by assure map tool. but when i read the map to the software, the fault occured.

hatem-darweesh, 2020-06-25:
@popeyedpanther, @robinchen2019  Sorry for the late reply, 
You can solve this problem by providing origin GPS data at the "Mapping Options" dialog while loading the map. 
for your map example. I just used first GPS point from the .osm file 

Lat: 29.6455991507
Long: -82.35307914587

and loading works fine.

popeyedpanther, 2020-06-30:
I tried using the GPS coordinates you used on the dialog box that pops up before map selection and it  still leads to a seg fault. I made a short video: https://www.dropbox.com/s/asie8gr2nbpg50c/2020-06-30%2012-12-17.mp4?dl=0

hatem-darweesh, 2020-07-01:
check this video: 
https://youtu.be/0nt63WhVj0A

Can you try downloading the latest version from the tool. 

sometimes it is mission library or wrong version. 
please follow exactly the setup steps in:
https://github.com/hatem-darweesh/assuremappingtools/tree/ubuntu18.04_build

popeyedpanther, 2020-07-13:
I performed a fresh install on a new VM and now it loads the lanelet2 files without crashing.

closed 2020-07-13 by popeyedpanther