hatem-darweesh/assuremappingtools #17: Please update to opencv 3.2.0 version or above

FajunChen, 2020-07-08:
It is not easy to install opencv 2.4 on Ubuntu 1806. If possible could you please update your opencv vesion to a the default 3.2.0 version or above? Thank you for you contribution.

hatem-darweesh, 2020-07-25:
Did you tried the branch https://github.com/hatem-darweesh/assuremappingtools/tree/ubuntu18.04_build
$ git checkout ubuntu18.04_build

FajunChen, 2020-07-25:
Thank you for your reply. There is no problem after checking out the ubuntu1804_build branch. 

I have another problem then. Could this tool extract  lanelet2 infomation from point cloud flie? From your tutorial video, I see that your lanelet2 file is loaded by hand, which is not converted from the pointcloud file. Thank you for you contribution.

hatem-darweesh, 2020-07-25:
PointCloud map doesn't have Road network map information. not lanelet2 or any other HD map format
but what happen is we manually draw the map and save it to any format we wish. 
in assure mapping tools there are some automation functions that can help make the map creation process faster.

FajunChen, 2020-07-25:
Could you please give me some promption about how to use those automationi functions? Or do you have any examples? Thanks.

hatem-darweesh, 2020-07-25:
check youtube tutorials https://youtu.be/lignj-gVGVg

FajunChen, 2020-07-25:
Cool, thanks. Do you have any plan to open the sources of your tools? Maybe someone others could improve it with you together, providing things such as better UI.

hatem-darweesh, 2020-07-26:
Thank your suggestion, maybe in the future.

closed 2020-07-26 by hatem-darweesh