hatem-darweesh/assuremappingtools #19: failed to find elevation tag for node and no visualization in RVIZ

TheGermanKid, 2020-07-14:
Howdy, 

first of all thank you for this great tool - making the maps themselves is a breeze compared to any other tool I have used thus far. 

I am running the latest autoware version available on github. 

I do have a problem which is probably user error - when i run the lanelet2_validation node on the created lanelet2 map i get the following error: "failed to find elevation tag for node xxxx:" i used the function "fix height" on all items but this did not resolve anything. Not sure what I am doing wrong with that. 

I tried to just upload the map to autoware but it will not show up in RVIZ whatsoever. Probably linked to the issue above but no sure. I also tried to use all options for the projection (no projection, MGRS and UTM) with no luck. 

Additional question: what does Lane Action mean and what does R L F stand for? 

Thank you!

hatem-darweesh, 2020-07-15:
Hello, 

Thank you for using ASSURE tools, and thanks for the feedback. 
I will check this, but I need some help from you. 
1- the .osm file
2- how do you validate is with lanelet2_validation
3- what Autoware version are you using. 

About your second question: 
the actions is used by OpenPlanner's global planner to optimize for the shortest route. R -> going right, L-> going left, F -> Moving forward. 
for example, in an intersection, if you assign very high cost to with R. then the global planner will try to avoid using this lane unless it is the only way to the goal position. 

Regards,

TheGermanKid, 2020-07-15:
1. Two files are attached below - obioulsyWrong.txt is very simplified because I was trying different things. I attached the .osm as text file as github won't let me upload .osm. I realize that there are no lon/lat or elevation data attached on this one. On previous ones (that had other errors besides) all was filled out which i attached as OtherErrors.txt. It gave me the same error but also other errors because I messed up the lanes.

[OtherErrors.txt](https://github.com/hatem-darweesh/assuremappingtools/files/4922347/OtherErrors.txt)
[obviouslyWrong.txt](https://github.com/hatem-darweesh/assuremappingtools/files/4922348/obviouslyWrong.txt)

2.I tried to validate by running: 
> rosrun lanelet2_validation lanelet2_validate  _map_file:=/pathtofile

3.according to my src/common git describe --tags it is 1.14.0-alpha which i actually got from when autoware.ai was still on gitlab. 
 
Thank you for answering that! Makes a lot of sense. 

My visualization issue may also be an issue with my projection. I am located in Texas, USA if that helps you debug this issue.

hatem-darweesh, 2020-07-15:
I checked with the .osm viewer.  the map origins is at (0,0,0). 
I tried with assure but it didn't show up. 
I have some questions: 
1- how this map is originally created ? 
2- if you are using point cloud as base to create road network map, then you need GPS origin to export to lanelet2 .osm map. 

before even we go to the validation check, we should be able to view the map in assure & JSOM tools

TheGermanKid, 2020-07-15:
I see, I tried two different ways:

1. Upload .pcd (created with NDT_Mapping in autoware) to Assure, then create lane, lines etc. 
2. Use TierIV mapping tool, upload .pcd then create lane. Tried to take that .osm but cannot upload to assure because the reference grid was invalid for what I was doing. 

As far as origin goes, this is not just the first value in the pointcloud?  I looked up some stuff for pcl and they have an origin calculator so i figured i need to do that probably. Here is the latest map I had that didn't give me height issue but i was messing around with the lane_direction tag. 

I attached below the .osm and the .osm.dat
`<?xml version="1.0"?>
<osm version="0.6" generator="lanelet2">
  <node id="2496" visible="true" version="1" lat="-0.00100860711" lon="-0.00071209634">
    <tag k="ele" v="-3.3532694816589355" />
  </node>
  <node id="2497" visible="true" version="1" lat="-0.00103049636" lon="-0.00070656264">
    <tag k="ele" v="-3.3532694816589355" />
  </node>
  <node id="2498" visible="true" version="1" lat="-0.00104632131" lon="-0.00085928898">
    <tag k="ele" v="-3.680045353062106" />
  </node>
  <node id="2499" visible="true" version="1" lat="-0.00106821056" lon="-0.00085375527">
    <tag k="ele" v="-3.680045353062106" />
  </node>
  <node id="2500" visible="true" version="1" lat="-0.00104844846" lon="-0.00086756264">
    <tag k="ele" v="-3.6992685520469824" />
  </node>
  <node id="2501" visible="true" version="1" lat="-0.00107052039" lon="-0.00086279838">
    <tag k="ele" v="-3.6992685520469824" />
  </node>
  <node id="2502" visible="true" version="1" lat="-0.00105978372" lon="-0.00091994199">
    <tag k="ele" v="-3.9194584587737591" />
  </node>
  <node id="2503" visible="true" version="1" lat="-0.00108186847" lon="-0.00091523675">
    <tag k="ele" v="-3.9194584587737591" />
  </node>
  <node id="2504" visible="true" version="1" lat="-0.00106166056" lon="-0.00092863323">
    <tag k="ele" v="-3.956366033879152" />
  </node>
  <node id="2505" visible="true" version="1" lat="-0.00108376433" lon="-0.00092401695">
    <tag k="ele" v="-3.956366033879152" />
  </node>
  <node id="2506" visible="true" version="1" lat="-0.00106365031" lon="-0.00093795646">
    <tag k="ele" v="-3.9951570760967727" />
  </node>
  <node id="2507" visible="true" version="1" lat="-0.00108547595" lon="-0.00093218019">
    <tag k="ele" v="-3.9951570760967727" />
  </node>
  <node id="2508" visible="true" version="1" lat="-0.00106667467" lon="-0.00094850518">
    <tag k="ele" v="-4.0431492524966792" />
  </node>
  <node id="2509" visible="true" version="1" lat="-0.00108707912" lon="-0.00093888325">
    <tag k="ele" v="-4.0431492524966792" />
  </node>
  <node id="2510" visible="true" version="1" lat="-0.00107064775" lon="-0.00095681006">
    <tag k="ele" v="-4.1207278978316175" />
  </node>
  <node id="2511" visible="true" version="1" lat="-0.00109080477" lon="-0.00094668663">
    <tag k="ele" v="-4.1207278978316175" />
  </node>
  <node id="2512" visible="true" version="1" lat="-0.00107475217" lon="-0.00096487242">
    <tag k="ele" v="-4.2033395386992378" />
  </node>
  <node id="2513" visible="true" version="1" lat="-0.0010948174" lon="-0.00095457065">
    <tag k="ele" v="-4.2033395386992378" />
  </node>
  <node id="2514" visible="true" version="1" lat="-0.00107813557" lon="-0.00097100538">
    <tag k="ele" v="-4.2875073359648628" />
  </node>
  <node id="2515" visible="true" version="1" lat="-0.00109969364" lon="-0.00096431072">
    <tag k="ele" v="-4.2875073359648628" />
  </node>
[... 795 lines cut ...]
  <relation id="2561" visible="true" version="1">
    <member type="way" ref="2550" role="left" />
    <member type="way" ref="2549" role="right" />
    <tag k="subtype" v="road" />
    <tag k="type" v="lanelet" />
  </relation>
  <relation id="2604" visible="true" version="1">
    <member type="way" ref="2563" role="left" />
    <member type="way" ref="2562" role="right" />
    <tag k="subtype" v="road" />
    <tag k="type" v="lanelet" />
  </relation>
  <relation id="2665" visible="true" version="1">
    <member type="way" ref="2606" role="left" />
    <member type="way" ref="2605" role="right" />
    <tag k="subtype" v="road" />
    <tag k="type" v="lanelet" />
  </relation>
  <relation id="2666" visible="true" version="1">
    <member type="way" ref="2715" role="outer" />
    <tag k="subtype" v="freespace" />
    <tag k="type" v="multipolygon" />
  </relation>
</osm>
`

hatem-darweesh, 2020-07-15:
you see these are wrong (latitude, longitude) values. 
I guess you need to consider the GPS origin when creating the point cloud map, then the pipeline will work. 
if you use the .pcd in assure maps, you can create .kml map and use it with Autoware. (but in this case you will use only OpenPlanner). 
or export .csv files from TierIV mapping tools and avoid using lanelet2.

TheGermanKid, 2020-07-15:
Yes i see. I will try some different things and probably get new data as well. Thank you a lot for your kind help! Instead of using my own data I will probably try the sample dataset first!

TheGermanKid, 2020-07-15:
this up to the first line of my .pcd file i used:
`# .PCD v0.7 - Point Cloud Data file format
VERSION 0.7
FIELDS x y z intensity
SIZE 4 4 4 4
TYPE F F F F
COUNT 1 1 1 1
WIDTH 3181304
HEIGHT 1
VIEWPOINT 0 0 0 1 0 0 0
POINTS 3181304
DATA ascii
57.905525 -103.08224 -16.56731 63`

hatem-darweesh, 2020-07-15:
Sorry , I never used the NDT mapping. or checked the details of .pcd file. 
my best guess it is auxiliary data in the process of creating the map and later in the process of creating the vector map.

TheGermanKid, 2020-07-15:
No worries! Thanks again for your help. If i get a workaround I'll let you know

closed 2020-09-12 by hatem-darweesh