hatem-darweesh/assuremappingtools #20: undefined symbol: _ZN10PlannerHNS11RoadNetwork13g_max_curb_idE

xqrkushnir, 2020-07-20:
Hi! I'm getting this error when trying to launch assure_map editor:
```
me: ~/Desktop/assuremappingtools$ ./assure_map_editor 

./assure_map_editor: symbol lookup error: ./assure_map_editor: 
undefined symbol: _ZN10PlannerHNS11RoadNetwork13g_max_curb_idE
```
What can cause it? 
Regards, Andres

hatem-darweesh, 2020-07-20:
It seems you have mixed library versions. 
please clean the lib directory , pull again and use export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:libs

xqrkushnir, 2020-07-20:
Unfortunately, that didn't help. 

I've reinstalled and rebuilt OpenCV 2.4, removed and installed back all the dependent packages, and did "_export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:libs_". Also, restarted my PC, just in case.

hatem-darweesh, 2020-07-20:
This issue has nothing to do with OpenCV. 
op_planner.so is wrong version. 
can you check after cloning that you are on the correct branch.

xqrkushnir, 2020-07-20:
Sure!

```
me:~/Desktop/assuremappingtools$ git fetch 
me:~/Desktop/assuremappingtools$ git status
On branch ubuntu18.04_build
Your branch is up to date with 'origin/ubuntu18.04_build'.

nothing to commit, working tree clean

me:~/Desktop/assuremappingtools$ export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:libs
me:~/Desktop/assuremappingtools$ ./assure_map_editor 
./assure_map_editor: symbol lookup error: ./assure_map_editor: undefined symbol: 
_ZN10PlannerHNS11RoadNetwork13g_max_curb_idE
```

xqrkushnir, 2020-07-20:
The issue covers somewhere within a conflicted package with Autoware.
With removing sourcing of Autoware's setup.sh from my **.bashrc**, I can launch the software without any issues.

The line I commented was this one:
``` source ~/autoware/devel/setup.bash ```

Just having ``` source /opt/ros/melodic/setup.bash ``` does the trick.

closed 2020-07-20 by xqrkushnir