hatem-darweesh/assuremappingtools #21: Linking errors at Ubuntu 18.04

mschratter, 2020-07-22:
Dear Hatem,

Thanks for your tool! It is very helpful for me, I used it previously with Ubuntu 16.04.

I switched my PC to Ubuntu 18.04 and use now the ubuntu18.04_build branch. I installed the specific OpenCV version, the other required libs and added export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:libs.

When I try to start the editor, I get following output:
**./assure_map_editor: symbol lookup error: ./assure_map_editor: undefined symbol: _ZN7lanelet10LaneletMapD1Ev**

I tried it out on a second computer with a similar behavior:
**./assure_map_editor: symbol lookup error: ./assure_map_editor: undefined symbol: _ZN10PlannerHNS11RoadNetwork13g_max_curb_idE**

Do you have a hint what is missing?

Best, Markus

tekrajchhetri, 2020-07-22:
@mschratter  this happens when there is dependency conflict. There was similar issue with me as well and the clean dependency management solved the issue and should works for you as well.

hatem-darweesh, 2020-07-22:
Thank you for using ASSURE tools, 
please check issue #20

mschratter, 2020-07-22:
I have seen the old issue, but I tried it out without success. I have a clean Ubuntu installation, no Autoware.AI is installed and only ROS is sourced (source /opt/ros/melodic/setup.bash).
@tekrajchhetri How did you solve the dependency issues?

tekrajchhetri, 2020-07-22:
@mschratter  can you share what you have in your path and ld library path variable?

mschratter, 2020-07-22:
@tekrajchhetri I have following in the paths:
markuss@markuss-X299:~/mapping/assuremappingtools$ echo $PATH
/opt/ros/melodic/bin:/home/vifware/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin

markuss@markuss-X299:~/mapping/assuremappingtools$ echo $LD_LIBRARY_PATH
/opt/ros/melodic/lib:/home/vifware/mapping/assuremappingtools/libs

tekrajchhetri, 2020-07-22:
I am not sure what you have in _/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin_ path but its something uncommon I am seeing. What I would suggest is to temporarily only include ros and the dependencies path related to assure tools. This way you can be sure there is no conflict, and it should hopefully solve your issue. If not then there should be some other issue.

So the question might be what would I do of other path information. Clearly they all are not used at a time. So better to activate them only when you need a simple bash script, a preferred way.

hatem-darweesh, 2020-07-23:
Can you search your hard drive for op_planner.so , how many copies do you have ?

mschratter, 2020-07-24:
@hatem-darweesh I find the libop_planner.so in your tool and in Autoware. But Autoware is not sourced by default.

markuss@markuss-X299:/$ locate op_planner.so
/home/markuss/mapping/assuremappingtools/libs/libop_planner.so
/home/markuss/ros/autoware.ai/build/op_planner/devel/lib/libop_planner.so
/home/markuss/ros/autoware.ai/install/op_planner/lib/libop_planner.so

hatem-darweesh, 2020-07-24:
can you remove the build folder of Autoware then try to run assuremappingtools again ?

mschratter, 2020-07-24:
Hi, do I need to install the op_planner separately?

```
markuss@markuss-X299:~/mapping/assuremappingtools$ export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/markuss/mapping/assuremappingtools/libs
markuss@markuss-X299:~/mapping/assuremappingtools$ echo $LD_LIBRARY_PATH
/opt/ros/melodic/lib:/home/markuss/mapping/assuremappingtools/libs
markuss@markuss-X299:~/mapping/assuremappingtools$ locate op_planner.so
/home/markuss/mapping/assuremappingtools/libs/libop_planner.so
markuss@markuss-X299:~/mapping/assuremappingtools$ ./assure_map_editor 
./assure_map_editor: symbol lookup error: ./assure_map_editor: undefined symbol: _ZN7lanelet10LaneletMapD1Ev
```

hatem-darweesh, 2020-07-25:
please try empty Ubuntu 18.04 installation without Autoware. only ROS

mschratter, 2020-07-26:
I was on the wrong branch. Sorry for the circumstances. @hatem-darweesh @tekrajchhetri  Thanks a lot for your help!

closed 2020-07-26 by mschratter

hibetterheyj, 2020-07-29:
Hello, my platform is 18.04 with zsh. I have downloaded and used your amazing tool 5 days ago. However, I get the same output today:
```
./assure_map_editor: symbol lookup error: ./assure_map_editor: undefined symbol: _ZN7lanelet10LaneletMapD1Ev
```

The repo status is as follows:
```
➜  assuremappingtools git:(ubuntu18.04_build)
```

`PATH` and  `LD_LIBRARY_PATH` is as follows:
```
➜  assuremappingtools git:(ubuntu18.04_build) echo $PATH
/opt/ros/melodic/bin:/home/heyujie/.local/bin:/home/heyujie/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
➜  assuremappingtools git:(ubuntu18.04_build) echo $LD_LIBRARY_PATH
/opt/ros/melodic/lib:/opt/ros/melodic/libibs:/home/heyujie/Softwares/assuremappingtools/libs
```

Thanks in advance !