hatem-darweesh/assuremappingtools #23: How to view the real ground at a customized z range

FajunChen, 2020-07-26:
When I select the "view ground" menu, all cloud points disappear. I think the real ground plane should be located at some z range. Could I customize where the ground plane is?

FajunChen, 2020-07-26:
When I want to add a new lane, I hope the z value is set on the proper value. Currently, I guess the z value is default to 0.  When I want to draw a new lane, I can't see clearly where the new lane goes, since it would be covered by the thick point clouds.

FajunChen, 2020-07-26:
It is best to provide some shader filter such as EDL, thus to better understand structures in the point clouds.

hatem-darweesh, 2020-07-26:
You are correct, when you are drawing, yes, z is zero, and drawing is not clear sometimes because of the point cloud. Thanks for the tip. 
but after you finish drawing the lane , I try to set the Z value with the closes z  value from the point cloud. 
also you can select "fix height" , it does the same.

FajunChen, 2020-07-26:
So how to find the most proporiate Z value? Do you ever consider to use a lane detection algorithm, and keep only related point cloud when drawing a vector map by hand?

hatem-darweesh, 2020-09-12:
The automation tools should take care of this issue, for now you need to adjust the z manually.

closed 2020-09-12 by hatem-darweesh