hatem-darweesh/assuremappingtools #26: OpenDRIVE maps import issue

mpechinger, 2020-07-28:
First of all, thanks for providing the possibility to convert OpenDRIVE maps into a compatible format to be used in autoware.ai! I want to kindly ask you, if you can have another look into the problem described in #10

I have done as you have said in this issue, but I am still not able to resolve the problem. What I have tried so far:

- Installing assure mapping tools on Ubuntu 16.04 step by step strictly following the instruction steps one to four.

- I did this procedure twice to make sure I get the same result and didn't do anything wrong by accident. 

- I have applied the same procedure for the Ubuntu 18.04 installation. Following the steps and installing it twice.

- Every installation was performed on a fresh Linux installation in a virtual machine. The files I have used to test the functionality of the import tool were the sample files from opendrive.org. 

Since this is the only tool, I'm aware of, which provides the interface to OpenDRIVE maps I want to reopen the discussion on this problem I am experiencing. Thanks for your hard work, I really appreciate it!

[image]

hatem-darweesh, 2020-07-28:
Sorry, I have to comment similarly. 
I couldn't reproduce this issue on any of my computers neither virtual machine.
but I will try to provide a docker image soon, so you don't have to install anything.

mpechinger, 2020-07-29:
Thanks for the quick reply. A docker image would be really nice!

closed 2020-07-29 by mpechinger