hatem-darweesh/assuremappingtools #27: Problems stitching lanes and recognizing adjacent lanes

abhi5691, 2020-09-02:
Hello,

Thanks for this awesome looking tool. However for someone new to Lanelet2 format I am having my share of trouble adopting them into AW. I cloned your [fork](https://gitlab.com/hatem-darweesh-autoware.ai) for autoware such that OP can now use Lanelet2 data however map creation using assuremappingtools is not so straightforward.

Each lane I draw in the assuremappingtools stays a standalone entity and the planner fails to plan a path across them (however close they be, even if I enable lane change). Similarly enabling lane change doesn't help OP to generate a path that involves LaneChange. This was very efficient with the vector maps that I have been using until now.

Can you indicate what might be missing/wrong here @hatem-darweesh ?

Further, However many other entities I may draw in the map (crossing/intersection/lines etc.) I don't see any of them in rviz upon import to AW. I only see the lanes. Is that normal?

Thanks

hatem-darweesh, 2020-09-04:
Hi @abhi5691 , 
you need to connect the lanes you create in assure tool. 
please check the tutorials: 
https://www.youtube.com/playlist?list=PLVAImlqqGbr4G8GXMsqp6dMhbKxTCEnrh

In the Info View. when selecting lane , add "From Lanes" and "To Lanes" Ids 
At the last waypoint in the lane you need to set "To Points" Ids 
At the first waypoint in the lane you need to set "From Points" Ids 

Hope this is helpful.

abhi5691, 2020-09-07:
It helps indeed @hatem-darweesh . Thanks for sharing this.

Can you also kindly shed some light on the second part? Shouldn't visualization be inclusive of other map features (_stop-lines_, _lines_, _curbs_, _crossings_, etc.?) I see nothing but the **lanes** (with a drive-able area around) as shown in the original image upload. I am just switching to lanelet2 and know only little about it, so any help from you will go a long way. Please let me know how I can ensure that the information I am feeding into the map build via this tool is actually getting to Autoware when the map is loaded.

hatem-darweesh, 2020-09-12:
It depends on the application, 
when running a node such as traffic light detection, if detection works then you have correct traffic light information in the map, if not, then you need to find where is the problem.

closed 2020-09-12 by hatem-darweesh

harderthan, 2021-01-14:
> Hi @abhi5691 ,
> you need to connect the lanes you create in assure tool.
> please check the tutorials:
> https://www.youtube.com/playlist?list=PLVAImlqqGbr4G8GXMsqp6dMhbKxTCEnrh
>
> In the Info View. when [...]

Hello, @hatem-darweesh .
From your reply, how to set Ids on "From Points" or "To Points" ?
I saw "editing lanes" youtube video. There is no tip to create ID section on  it.

I did double-click and one tab, ctrl+click and the like...

hatem-darweesh, 2021-01-14:
@harderthan 
For example to connect lanes: 
- select the lane 
- In the information view click on "To Lanes" 
- use keyboard "+" , that will insert new id with value 0 
- change to value to the id of the next lane 
- when you still focus on the Information view press "ctrl + s" 

If you want to delete a connection, click on the id and press "Delete" button 

In the repository under "Docs" folder you will fin new documentation file, please check it.