hatem-darweesh/assuremappingtools #28: How to use MGRS projection to generate the autoware extension lanlet2 map

jackfine, 2020-09-08:
Hello, Thank for this awesome tool !
I have met some problems when I was trying to import  the .osm map into autoware.

The first problem comes out when I was using UTM projection to save the lanelet2 map. 
I only draw the lane and the .kml map was successfully showed in google earth . But I can't not import the lanelet2 map into autoware , the problem autoware reported is "Uninitialized quaternion ,assuming identity." 
The .osm file has the right latitude and longtitude information , but there was no "turn tag" information which was Mandatory tag in autoware document .

The second question comes out when I was using MGRS projection to generate the lanlet2 map. I was wondering whether the projection string was the same as mgrs code . I use the mgrs code according to my location and input the latitude and longitude information . But  .osm file has no latitude and longitude data  when I saved the lanlet2 map .

hatem-darweesh, 2020-09-12:
@jackfine 
if you can share the .osm map , I will check whey it is not working with Autoware.

jackfine, 2020-09-14:
Thanks for your help! This is the lanelet2.osm map.  [https://github.com/jackfine/lanetlet2]

hatem-darweesh, 2020-09-14:
I don't know which Autoware version are you using. 
but the map seems fine.

jackfine, 2020-09-14:
I'm using Autoware 1.14,  Ubuntu18.04, ROS melodic. Should I try another autoware version? Which version of autoware do you recommend？
I have another question when I'm open .kml map in google_earth. I generate the .osm map based on NDT point cloud, but the point cloud‘coordinate is unable to adjust in the assuremappingtool. So the .kml map can not accurately match the map in reality. Is there an method to adjust the point cloud'coordinate that I have't found?

hatem-darweesh, 2021-01-12:
You need to fix the coordinate relation between point cloud map and the HD map, both must have same GPS origin. 
currently you can use the generated map with any Autoware version above 1.11

closed 2021-01-12 by hatem-darweesh