hatem-darweesh/assuremappingtools #31: How to add lane_change flag in .kml when converting .osm to .kml

kyujinChoi, 2020-10-10:
Hello @hatem-darweesh !
I am making my own ADAS map to use in Autoware.

- Steps:
1. Extracted lanes, stop lines, crosswalk, traffic lights from points map.
2. Converted file format(**.pcd -> .osm**)
3. Make lanelet and added tags in linestrings(ex. (_subtype=dashed, type=line_thin, lane_change=yes_) or (_subtype=solid, type=line_thick_)) in [josm](https://josm.openstreetmap.de/) editor refer to [lanelet2 documentation](https://github.com/fzi-forschungszentrum-informatik/Lanelet2/blob/master/lanelet2_core/doc/LinestringTagging.md) and [Lanelet2 format for Autoware](https://gitlab.com/autowarefoundation/autoware.ai/common/blob/a82142cb990b98a62e14fb68ed0280321c368dd4/lanelet2_extension/docs/lanelet2_format_extension.md).
4. Converted file format(**.osm -> .csv**) using [lanelet_aisan_converter](https://github.com/Autoware-AI/utilities/tree/master/lanelet_aisan_converter), but seems not supporting some fields in [stop_line.csv](https://github.com/Autoware-AI/utilities/blob/master/lanelet_aisan_converter/lib/lanelet2aisan_converter/create_feature.cpp). I want to link _stop_line_ and _traffic_light_.
5. So I tried to convert file format(**.osm -> .kml**) using [assuremappingtools](https://github.com/hatem-darweesh/assuremappingtools/), but does not change lane in op_planner 2.0.

I want my ADAS map to contain _lanes(solid and dashed), stop_line, traffic_light and light_bulbs_.  Do you have any idea for this? 

BTW I appreciate for your tools :D

Kyujin Choi

hatem-darweesh, 2020-10-11:
Hi, Thank you for using ASSURE mapping tools. 

I have some question first: 
1- Does the lane change work in any way ? 
2- Which map do you like to use (kml, AISAN, lanelet2) ? 
3- if you are using OpenPlanner 2.0 , just create the map using ASSURE tool. 

Currently you need to check the connection between the traffic light, stop line, and closet lane waypoint manually. 
check the information you need to fill in "info view" for each map item. 

In the latest ASSURE tool , there is an automatic way function that detects the parallel lanes and set the lane change information automatically. 

Regards,

kyujinChoi, 2020-10-11:
Thank you for the replying.

1. Lane change work fine when using AISAN format(lanelet_aisan_converter). However kml does not. Maybe because I am not familiar with ASSURE mapping tools. Is there any documentation? or example kml file? I already watched your tutorials on youtube, but cannot find how to set the lane change information manually.
2.Both kml and AISAN are fine. But AISAN doesn't support stop_line now. So I prefer kml more now.

Can I get documentation for how to connect Ids between light-lane-stopline also?

hatem-darweesh, 2020-10-11:
I see, 
it is a new function, that is why it is not in the tutorial video. 
but this function is active by default when you load AISAN map, because AISAN map format doesn't have lane change information. 
When you load the map into ASSURE Tool, 
1- right click 
2- Functions 
3- Find Side Lanes 
4 save the .kml map. 

This should work. 

Regards,

closed 2020-10-12 by kyujinChoi