hatem-darweesh/assuremappingtools #33: Consider Hosting Docker Containers?

JWhitleyWork, 2020-10-23:
Would you be interested in hosting the Docker container that is built by `build-docker.sh` in a Github Docker Repository or DockerHub? Building the container is very time-consuming given the number of dependencies and pulling it from a repository would be significantly faster.

hatem-darweesh, 2020-10-28:
Hi @JWhitleyWork , thanks for your suggestion. 
The docker was a quick fix for the dependencies problems. 
The ultimate goal is to host the first stable release version. I guess I can do this for the experimental versions after coming back from my vacation.

Aladdin-Ark, 2021-01-11:
The problem for hosting on Docker-Hub is related to NVIDIA driver.
It is very likely to not work on different systems, and it would be required to build the image again locally and obtain NVIDIA driver based on the driver that is used on the host.
Another solution is to provide the base image with all the required libraries on Docker-Hub and just provide a build script that is using the base image to build NVIDIA driver inside it, this should not take long to build as the current version.

hatem-darweesh, 2021-01-12:
Thanks @Aladdin-Ark

closed 2021-01-12 by hatem-darweesh