hatem-darweesh/assuremappingtools #35: How to correctly load a lanelet2 map in Autoware?

salvatoresan, 2020-10-27:
Hi @hatem-darweesh, i'm currently experimenting on Carla and Autoware v1.14 and my goal is to generate an HD maps in the lanelet2 format starting from an OpenDRIVE file obtained from RoadRunner. Leaving aside a conversion issue due to some inconsistencies in the OpenDRIVE file, i couldn't even load in Autoware the official lanelet2 maps of Carla, which i assume they were created with the Assure mapping tools, in the map tab in Autoware whereas there weren't any issues with the vector maps. Besides, i noted that loading a pcd map and the corresponding lanelet2 or vector map causes a matching issue where the pcd map appears above the lanelet2/vector map. What would be the correct workflow to generate a lanelet2 map and then load it in Autoware? Could the issue be related to an error in  the type of coordinate system (UTM/MGRS) or in the tf launch file in the Carla-Autoware bridge?
In any case, thanks for your hard work

hatem-darweesh, 2020-10-28:
I tried using CARLA with .kml maps. 
OpenDRIVE -> KML -> OpenPlanner with CARLA. 
The lanelet2 map generation will be modified and tested next month. 
It should be ready by the end of November

hatem-darweesh, 2021-01-12:
Hi @salvatoresan , you can try the lanelet2 save option now.

closed 2021-01-12 by hatem-darweesh