hatem-darweesh/assuremappingtools #38: new script docker error

sethut, 2021-01-11:
[image]

with new script, zatitech/assuremappingtools not found error. maybe image and tag name is wrong

sethut, 2021-01-11:
> can you check the docker images you have

thanks. it just my fault

closed 2021-01-11 by sethut

ayberksener, 2021-01-13:
@sethut @hatem-darweesh 
I am getting the same error, can you please guide me how you resolve this issue. Thank you

sethut, 2021-01-16:
> @sethut @hatem-darweesh
> I am getting the same error, can you please guide me how you resolve this issue. Thank you

just run build script first
I misunderstood as docker pull is operating  when executing a run script

if you have an error, check the image
$docker images
if you don't have an image zatitech/assuremappingtools:master, docker image is not build yet
but if you have an image above, i don't know about that. that's not same issue

sethut, 2021-01-16:
@ayberksener 
just run build script first
I misunderstood as docker pull is operating when executing a run script

if you have an error, check the image
$docker images
if you don't have an image zatitech/assuremappingtools:master, docker image is not build yet
but if you have an image above, i don't know about that. that's not same issue