hatem-darweesh/assuremappingtools #41: PCP and lanelet2 map not aligned

MichaelW93, 2021-02-15:
Hello,

I am using Carla 0.9.10.1 and Autoware 1.14. 
I have currently updated your mapping tools to the newest version. Now I have the issue, that my point cloud and my vector map are not aligned anymore.
Do you know a quick fix for this issue?

I also have another problem. I don't know if you are the right person to ask but I am running out of ideas:
Instead of driving the red line, autoware follows the blue line, which makes no sense to me. I have checked all the lanes, if they are connected properly, redraw them by hand but nothing worked. 
The issue seems to be starting at the middle of the corner. Because if take the same starting point, and just let it drive to the beginning of the corner it works without problems.
The same problem also occurs at some other points of this map.
Do you have any idea what could be the reason for this? I am fairly new to this whole topic and am stuck at this point for quiet some time.
[image]

Much thanks in advance

hatem-darweesh, 2021-02-15:
Hello, 

I would need more information to be able to know the source of the problem. 
how did you end up with this vector map (is it converted using Autoware) ? and what is the original source ? 
how do you specify the origin of the vector map and the point cloud ? 

about the planning issue, are you using OpenPlanner ? which version ? 

Regards,

MichaelW93, 2021-02-15:
Hi,

The Map itself was built using RoadRunner. It was saved as .xodr for Carla. I then converted it using your tool to lanelet2 and then used autoware to convert it to the aisan format. All this was done with the OLDER version (before your latest update).
Origin was always set to 0, and I used the umts projection.

Now I have reloaded the .osm map to try to find a solution for my planning problem with the NEWEST version of your tool. Now the both maps are not aligned anymore. When comparing the old and new .osm file, the "lat" and "lon" entry have changed.

For Planning I am using the op_global_planner from Autoware 1.14.
I hope these information help.

hatem-darweesh, 2021-02-15:
you can still use the old version , you will find tags 0.5 or 0.1 , or older branches. 
if you can share both the old and new versions that would help me find what is the problem. 
also I will check the map connectivity 
I will need the .xodr, .osm (old and new). 
if possible the point cloud. 
you can share the data via mail for privacy, and I will delete it after testing.

MichaelW93, 2021-02-16:
Hi,

I sent you a mail with the map files.
Thanks a lot!

closed 2021-05-18 by hatem-darweesh