hatem-darweesh/assuremappingtools #43: about lane merge and delete error lane

TerryLSR, 2021-02-26:
hello,
my ASSURETOOLS ver is 0.5, but it can not delete the error lane. The DELETE button is not useful. 
And if I want to merge one lane to anther lane, the value of "to Points" or " to Lane" is can  not be setted.
 Is there  some way to solve these problems?
[image]

Thanks.

hatem-darweesh, 2021-02-26:
To delete lane you select the lane (not the waypoint) , then press Ctrl + d + d 
to add item in the Info View: 
Add: '+' 
Delete: 'Delete' button

TerryLSR, 2021-03-01:
The operation that you asid to delete lane is useful ! thanks.
but how to merge one lane to anther lane, the value of "To Points" or " To Lane" is can not be setted.

hatem-darweesh, 2021-03-01:
Can you show me example by recording a video.

athulg, 2021-03-09:
> The operation that you asid to delete lane is useful ! thanks.
> but how to merge one lane to anther lane, the value of "To Points" or " To Lane" is can not be setted.

I've the exact same issue, I'm unable to manually enter any values in the "To Lanes" or "From Lanes" value box. The delete operation worked perfectly.

Thanks

hatem-darweesh, 2021-03-11:
Sorry I can not understand the issue with editing from a picture.

closed 2021-05-18 by hatem-darweesh