hatem-darweesh/assuremappingtools #44: error while converting in autoware from .osm to aisan

AthulGangadharan, 2021-03-01:
Hello, thank you for the amazing tool, I really appreciate your effort. 

I'm trying to convert the RoadRunner's .xodr to aisan format and I followed these steps

"1) Load the .xodr 2) Open Projection Information dialog 3) Remove the projection string 4) Save as .osm (Lanelet2) 5) Convert using the Autoware converter to vector map (use lat=0 and lon=0 otherwise you will get an error)"

When i try to convert the osm file into aisan I get the below errors. Any suggestions on what I might be doing wrong?

`o get id 7099 from map
[ERROR] [1614585720.523750556]: 	- Error parsing primitive 7073: Failed to get id 7101 from map
[ERROR] [1614585720.523764239]: 	- Error parsing primitive 7073: Failed to get id 7103 from map
[ERROR] [1614585720.523778037]: 	- Error parsing primitive 7073: Failed to get id 7105 from map
[ERROR] [1614585720.523792766]: 	- Error parsing primitive 7073: Failed to get id 7107 from map.........`

Thanks a lot!

AthulGangadharan, 2021-03-01:
I changed the lat & lon to 58 & 12 respectively as that was the value in the .osm file. 

But now I get the below error when I convert from lanelet2 to aisan
`terminate called after throwing an instance of 'lanelet::RoutingGraphError'
  what():  Could not determine relationship between lanelets following lanelet 4894. Lanelets that are diverging and merging at the same time are not permitted.
Aborted (core dumped)`

hatem-darweesh, 2021-03-01:
This logical error caused by lanelet2 validation, 
it is acceptable in OpenDRIVE and KML maps but not in lanelet2. 
when there is a merging point in the map and then branching point, you need to connect them by another lane manually. or modify your original .xodr map. 

Regards ,

AthulGangadharan, 2021-03-03:
Thank you for the info, [https://github.com/fzi-forschungszentrum-informatik/Lanelet2/issues/51](url) it seems like they've updated it in Lanelet2. Closing this issue for now.

closed 2021-03-03 by AthulGangadharan