hatem-darweesh/assuremappingtools #45: Generated Lanelet OSM map missing mandatary "turn_direction" tags.

privvyledge, 2021-03-14:
Thanks for this tool. 

I exported an OSM file from "https://www.openstreetmap.org" and exported it to OpenDRIVE format using MATLAB drivingScenarioDesigner. I was able to load it into "assuremappingtools" and saved it as a Lanelet2 OSM without errors. However, when validating and loading it into Autoware, I get the error "lanelet ****** seems to be intersecting other lanelet, but does not have turn_direction tagging" as shown in "Autoware_validation_error.png". 

[image]

What can I do to resolve this? 

Thanks.

hatem-darweesh, 2021-03-18:
Can you tell me more information, 
which version of Autoware, assure tool ? 
can you share the open drive map with me for testing ? 
also what happen when you load the converted lanelet2 using assure mapping tool ?

privvyledge, 2021-03-19:
### 1. What version of Autoware and assuremappingtools were used?
Autoware: current master branch as of this date.
Assuremappingtools: current master branch as of this date.

### 2. Can I share the OpenDRIVE map?
Yes. Where should I send it to?

### 3. What happened when you load the converted lanelet2 using assuremappingtools?
The generated lanelet2 map loaded on assuremappingtools without errors.

hatem-darweesh, 2021-03-20:
Send it to hatem.darweesh@gmail.com
both the OpenDRIVE and .osm 
I guess this Autoware validations issue. but I want to make sure.

privvyledge, 2021-03-20:
I have sent them to your email.

hatem-darweesh, 2021-03-23:
I was a bit busy recently, I will check the maps in the next couple of day.

hatem-darweesh, 2021-03-24:
Hi, 
I checked the OpenDRIVE Map and the .osm you have send me. 
I could load and visualize the .osm map into autoware. 
This is the error I get from Autoware: 
[ERROR] [1616584171.084392186]: Could not find valid vertex for ear clipping triangulation. Triangulation result might be invalid

but as general rule, the OpenDRIVE map is not a valid autonomous driving map, connectivity between lanes doesn't exist. 
and there is no smooth transition between roads. 

about direct conversion from open street map to lanelet2, there is no plan to do so. 
we will face lots of challenges as open street maps have many errors and missing lots of information , beside it is not accurate enough for autonomous driving

privvyledge, 2021-03-24:
1. How did you load and visualize it in Autoware?
2. How can you get the transforms from the map to the car?
3. What method do you recommend to then create a good enough 3D map of real-life locations (whether in a lab or on the road).

hatem-darweesh, 2021-05-18:
These questions are not related to this issue, 
please ask them on ros answers.

closed 2021-05-18 by hatem-darweesh