hatem-darweesh/assuremappingtools #46: Any plan to generate Autoware Lanelet2 OSM file from Open Street Map exported file directly.

privvyledge, 2021-03-20:
I have an Open Street Map file downloaded from https://www.openstreetmap.org/. 

1. Do you have any plan to load these .osm files (not Lanelet2)?
2. How can I generate a standard Lanelet2 .osm map from the Open Street Map .osm map?
3. How can I generate an Autoware compatible Lanelet2 .osm map?

Thanks for your help.

hatem-darweesh, 2021-03-23:
there is a converter from Open Street Map .osm file to OpenDRIVE. 
check this link 
https://sumo.dlr.de/docs/Tutorials/Import_from_OpenStreetMap.html

privvyledge, 2021-03-23:
I have tested that converter. I then loaded the OpenDRIVE map into assuremapping tools successfully and saved it as a Lanelet2 file. However, I kept getting the same error as shown in issue https://github.com/hatem-darweesh/assuremappingtools/issues/45. Thanks.

closed 2021-03-23 by privvyledge

felixf4xu, 2024-04-26:
> there is a converter

what's the name of the converter? I can't find OpenDrive related information in the page