hatem-darweesh/assuremappingtools #47: Using generated Autoware Lanelet2 map.

privvyledge, 2021-03-20:
I downloaded the sample Autoware lanelet2 map (I am running into problems with my own map. See https://github.com/hatem-darweesh/assuremappingtools/issues/45#issue-831223040) and loaded it into assuremappingtools then saved it as a Lanelet2 map. However, I have some questions about loading it: 
### 1. How can a lanelet2 map generated be visualized in RViz?
### 2. How can the global planner be used with this map to generate waypoints and velocity?
### 3. How can a car be localized in this environment? 
### 4. How can the map topic /map be published?
### 5. How can the /tf transforms be computed? Like the map to base_link and world to map transforms.

hatem-darweesh, 2021-03-24:
so , you want to know how to use Autoware ? 
please check the online documentation, tutorials and videos that has the answers. 
It will take long time and huge effort to answer these question.

privvyledge, 2021-03-24:
I have done so but the documentation is incomplete and does not directly provide these answers. I will ask them in the appropriate channel though.

Thanks.

closed 2021-03-24 by privvyledge