hatem-darweesh/assuremappingtools #48: Installation issues

yufeilongyuan, 2021-04-15:
My operating system is Ubuntu 16.04. Is the installation guide **master** or **Ubuntu_ 16.04_ build**？ Thank you.

hatem-darweesh, 2021-04-15:
If you use Ubuntu_16.04, you will get older version from the tool. 
to use the latest version, I recommend using the docker installer

yufeilongyuan, 2021-04-15:
I have made this mistake with you. I don't know how to solve it. Thank you
[image]

Maxcode19, 2021-04-15:
Hi, I have the same issue of **yufeilongyuan**. I'm working with **docker** in **Ubuntu 18.04**. Differently from **yufeilongyuan**, I do not have:

**libGL error: No matching fbConfigs or visual found
libGL error: failed to load driver: swrast**

I simply have the first 3 lines (that should be OK) and:

**freeglut (assure_map_editor): Unable to create OpenGL 1.0 context (flags 0, profile 0)**

The weird thing is that up to few months ago the application perfectly worked. I am pretty sure that no modification has been done to the system since that moment. Do you have any hint on how to solve it? Thank you for your help

yufeilongyuan, 2021-04-15:
Oh, although I haven't completely solved this problem, I know why. That is the NVIDIA driver. When I block the NVIDIA  driver, I can use it, although the performance is relatively low. Your problem must be that the NVIDIA driver has been updated, which leads to incompatibility. It's just my guess.

Maxcode19, 2021-04-15:
Hi! Thank you very much for the help. I have figured it out. The problem was related to the Nvidia drivers that, for some reason, were no more activated. I installed the NVIDIA-DRIVER.run file and now it works!

yufeilongyuan, 2021-04-16:
Congratulations on solving the problem

closed 2021-04-16 by yufeilongyuan