hatem-darweesh/assuremappingtools #49: assure_map_editor not running

kanakab, 2021-04-17:
Hello,

Thanks for creating this tool.

I am working on Ubuntu 18.04, ROS -melodic and with latest assuremaapingtools  with docker installation.

sh build-docker.sh - worked no issues
./run-docker - launches the GUI

I need to convert an opendrive file to lanelet

When I run the command: ./assure_map_editor, I am getting the folllowing error
./assure_map_editor: error while loading shared libraries: libGLEW.so.2.0: cannot open shared object file: No such file or directory
 
Please help me resolve this issue

hatem-darweesh, 2021-04-18:
Can you check that your docker build had no installation error.

kanakab, 2021-04-20:
The issue is resolved, I was not using the GUI properly earlier. Thanks @hatem-darweesh

closed 2021-04-20 by kanakab