hatem-darweesh/assuremappingtools #50: ASSURE mapping tool for latest requirements for autonomous parking demo

congphase, 2021-04-19:
This issue is originated from [Autoware.Auto's gitlab issue](https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto/-/issues/1018)

**Expectation**: a simple lane that can be used to set a start pose and a goal pose for ` lanelet2_global_path_planner(_nodes) `

**How to re-produce my issue**:
- **My map data**:
  - .pcd: [link](https://drive.google.com/file/d/1G0SDx1SaA2j2Vl2CyU0fqbj6wyKbBj4n/view?usp=sharing)
  - .osm: [link](https://drive.google.com/file/d/1BgCWB_RAWrDPKDiPszUxdlHSkuVfgczK/view?usp=sharing) 
  - ` map_publisher.param.yaml `, ` map_publisher_vehicle.param.yaml `, ` lanelet2_map_provider.param.yaml `, ` autonomoustuff_parking_lot_lgsvl.yaml `: [link](https://drive.google.com/drive/folders/1SDW11H1UKg87u-yF6srQss4eoogBsY1u?usp=sharing)
- **Error log**:

```
[ERROR] [lanelet2_global_planner_node_exe-14]: process has died [pid 23500, exit code -11, cmd '/home/anhnv/AutowareAuto/install/lanelet2_global_planner_nodes/lib/lanelet2_global_planner_nodes/lanelet2_global_planner_node_exe --ros-args -r __node:=lanelet2_global_planner_node -r __node:=lanelet2_global_planner_node -r __ns:=/planning -r HAD_Map_Client:=/had_maps/HAD_Map_Service -r vehicle_kinematic_state:=/vehicle/vehicle_kinematic_state'].
```

- **Reason**: parking settings missing, causing ` parking_id_list ` to be null.

- Current situation: Don't know how to create a simple lanelet2 map with parking settings with ASSURE mapping tool.
- 
Thank you, Mr. Hatem Darweesh

hatem-darweesh, 2021-04-20:
To understand this issue, 
you can't generate parking trajectory using the lanelet map? 
or you can't generate any trajectory ? 

what are the parking settings? can you share with me reference to those lanelet2 items. 
are you talking about drop pick up point and drop off point, and parking spot ?

congphase, 2021-04-20:
> you can't generate parking trajectory using the lanelet map?
or you can't generate any trajectory ?

--> The ` autoware_auto_avp_demo ` can't generate any trajectory with any start pose (2D estimation) and goal pose I set. It yelled the error log as my above comment.

> what are the parking settings?

--> I meant they are ` parking_spots `, ` parking_access `, ` ref_lanelet `, and other possible tags that make a lane capable of picking a start and pose goal on. Their map allows start and goal pose being set at parking spots, or, on a lane with "parking settings". That's what I meant about "parking settings". Looking at their [.osm file](https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto/-/blob/master/src/tools/autoware_auto_avp_demo/data/autonomoustuff_parking_lot.osm), there are so many settings that I don't know how to bring them in my map to fit with my pcd, and successfully generate a trajectory.

Can you just take a look at their ` .osm ` file, and my ` .pcd ` file. And help me with creating a very simple semantic map only for this area in the circle?
[image]

hatem-darweesh, 2021-04-22:
I am Sorry, 
This will need lots of modifications and reviewing lots of Autoware.Auto design documents. 
I will add it to the future development list. Maybe by the end of this year. 
Sorry again. 
My advice: 
Study the Autoware.Auto map requirements and use JOSM to create your map.

congphase, 2021-04-23:
That's a sad thing to hear. Anyway, thank you a lot for the warm welcome, Mr. Hatem.

Hope the next release of ASSURE mapping tool will attract a large number of users.

closed 2021-04-23 by congphase