hatem-darweesh/assuremappingtools #51: LaneLet2 map is loading up  appropriately

ParthAD3, 2021-04-30:
hello, 

i am loading my .xodr file and then saving it in .osm (Lanelet2) format, and  when i load that saved file the road lines are broken and the traffic lights (Green dots ) are out of the whole map, its seems to be broken, can you help with this issue ?

[image]
[image]
[image]
[image]
[image]

ParthAD3, 2021-04-30:
i seem to have fixed the issue of broken lane lines but i am unable to fix the traffic lights going out of the map on the left side (Refer last image )

hatem-darweesh, 2021-05-03:
Hi @ParthAD3 , This issue was fixed with the latest commit. 
git the latest from "master" and check again.

ParthAD3, 2021-05-03:
Hey, i seems to be now working still, as you can see i am in the git master branch
[image]

hatem-darweesh, 2021-05-04:
1- load the OpenDRIVE .xodr map 
2- remove the projection string from the map info dialog 
3- save to lanelet2 .osm map

ParthAD3, 2021-05-04:
> 1- load the OpenDRIVE .xodr map
> 2- remove the projection string from the map info dialog
> 3- save to lanelet2 .osm map

Yes, i have executed these steps just as mentioned, also i have un select the show all roads.
After saving, when i load the labeler.osm file to check it, it loads up as the above image.

hatem-darweesh, 2021-05-04:
You need to rebuild docker

hatem-darweesh, 2021-05-12:
@ParthAD3 were you able to get it to work ?

ParthAD3, 2021-05-14:
hey @hatem-darweesh 
so, i rubuilt the docker, now the updates are,
1. the traffic light are in their position now i.e. in the maps but i have 1 ques regarding the traffic lights : 
> what does the horizontal angle of the traffic light (the arrows) mean, is it the traffic light direction (the facing of the tfl) to the traffic or something else because when i loaded the .xodr map the horizontal directions are different from the lanelet2.osm map direction (refer image)
[image]
the .xodr traffic light horizontal position (i don’t understand their horizontal position please explain) also in the right side of the image the two parallel gray stopping lines seems a bit wrong than the others
[image]

2. in this lanelet2 map i don’t have the other road boundaries in the map which were there in the .xodr map, and it seems they were not present in the previous update aswell you can refer the image above(which was having the traffic light issue)  
> what i want is the white boundries line and the yellow middle lines as image below.
[image]

can these things  from point 2. be implemented then that’ll be perfect, and please explain my point 1. question. Thankyou

hatem-darweesh, 2021-05-14:
Can you make your question simpler and shorter, I don't understand what is the problem
for you information: 
- traffic light angle is wrong in the OpenDRIVE map 
- .xodr doesn't have stop line, only something called traffic light reference point, I use these points to create fake stop lines. 
- not everything in OpenDRIVE are supported in lanelet2 , so don't expect full conversion. 
-

ParthAD3, 2021-05-14:
ok, question:
1. what does the traffic light horizontal angle in the lanelet2 map mean. does it show the face of the traffic light  towards the traffic?
[image]
2. so, you mean to say that there wont be the white and the yellow lines from the .xodr converted in the loaded lanelet2 map ? because we needed those lane boundaries the yellow and the white outer ones, rest features we are ok with

hatem-darweesh, 2021-05-17:
1. Yes, it matches the traffic direction. even if the traffic light "yaw" in the OpenDRIVE is correct. lanelet2 map format does not support traffic light heading. (if you know otherwise please tell me). 
2. Sorry, the "lineStringLayer" which can represent lane markings in the lanelet2 map is not supported in the tool now, I will add it to the tasks list.

ParthAD3, 2021-05-18:
Basically, I could generate the Lanelet2 map and convert it to Vector Map using lanelet_aisan_converter. But the generated traffic light poles have a slightly different position comparing to the original ones and also all three lights arrows Blue, Yellow and Red are on the same position. So I can only see one.
Do you know what can I do to solve it?

closed 2021-06-27 by hatem-darweesh