hatem-darweesh/assuremappingtools #52: Issue opening assure

tanishafrancis, 2021-05-03:
I typed sudo ./run-docker but an error message appeared. Here is a piece of what I saw:

"Couldn't connect to accessibility bus: Failed to connect to socket /tmp/dbus-lbY2ttEZPq: Connection refused"

How do I fix this issue?

hatem-darweesh, 2021-05-04:
I don't know, this never happened to any of the users. 
Might be an issue with your docker installation, or linux drivers issues.

Yanlibao0518, 2021-05-09:
hello,
@hatem-darweesh 
when i typed sudo ./run-docker but an error message appeared:

”
No specific tag provided using 'master'
No specific workspace path provided. Running without workspace.
access control disabled, clients can connect from any host
./run-docker: line 67: docker: command not found
“
[image]

can you help with this issue ?

hatem-darweesh, 2021-05-12:
@Yanlibao0518 , you don't have proper docker installation. 
please make sure that you can pull and run default docker images. 
and then follow the assuremaptools readme instruction for using docker

closed 2021-06-27 by hatem-darweesh

BatuhanUlucay, 2021-08-02:
> I typed sudo ./run-docker but an error message appeared. Here is a piece of what I saw:
> 
> "Couldn't connect to accessibility bus: Failed to connect to socket /tmp/dbus-lbY2ttEZPq: Connection refused"
> 
> How do I fix this issue?

Error about /tmp/dbus is fixed when I purged and reinstalled my nvidia driver.