hatem-darweesh/assuremappingtools #54: Custom map origin

BatuhanUlucay, 2021-06-04:
I want to change the position of initial origin (or grid). Because when I load my pointcloud to the tool, I want to align my origin to somewhere I know. Is this possible?

Also is there any documentation about the projection section?

Thanks in advance

hatem-darweesh, 2021-06-07:
You will find new menu item called "shift map" which does exactly what you asked for. 
About projection, you will find only plenty of articles about projection coordinates, tools, libraries. 
In assuremappingtool I use libproj4 as the tool to calculate UTM from GPS and vice versa. 
also check the project readme.

closed 2021-06-07 by hatem-darweesh

BatuhanUlucay, 2021-07-02:
Thank you for the update! Do you mean translate map button with "shift map" ? It translates map items, but can I shift the pointcloud based on origin grid? Thanks.

hatem-darweesh, 2021-07-03:
Hi @BatuhanUlucay , similar function can be added. 
can you convince me why this function should be included. 
what I understand, if your point cloud map has wrong origin, you have to re-generate this map with accurate origin (pose + rotation).

BatuhanUlucay, 2021-07-05:
Hello again @hatem-darweesh 
I am importing my pointcloud, tool is putting a grid to the center of the pointcloud, but I cannot obtain the accurate position and rotation for that origin. I want to shift the whole map (and pointcloud) to the point where I know the position(my own origin).
Thanks for the quick answers!

hatem-darweesh, 2021-07-05:
Can you open separate issue for this function. and tell me what coordinates system are you targeting.

BatuhanUlucay, 2021-07-05:
[new issue](https://github.com/hatem-darweesh/assuremappingtools/issues/57)

congphase, 2021-07-17:
Hi all, is there any update on this?