hatem-darweesh/assuremappingtools #61: Problem while stitching lanes

BatuhanUlucay, 2021-09-06:
Hello Hatem,

I am trying to create an HD map for a custom territory, and I want to produce a lanelet2 map.
I encountered a problem while stitching more than two lanes. Altough I arrange "from" and "to" details for both lanes and waypoints. I cannot stitch three lanes at the same time. 

Could you please provide a solution or a documentation about this function?
I have watched your tutorials by the way, however I could not understand how stitching works.

Thanks in advance.

hatem-darweesh, 2021-09-06:
Stitch work automatically, and you need to load the map from kml and lanes must be connected. 
So, you have two separate lanes and you want to merge them into one lane ? 
you can use "Lane Functions" -> "Merge Next Lane"

BatuhanUlucay, 2021-09-06:
[image]

This picture is from an open source map named gomentum.
Above picture shows that connection waypoints of three lanes stand at the exact same position.  I believe that this is achieved by using stitch function. Am I right?

[image]

In my custom map, I want to do the same thing; however,  the waypoints do not fit appropriately. And I am not sure that the openplanner will follow without a connection(by stitching) between them.

Lastly, I wonder is the stitch function is mandatory? Or is it just optional?

hatem-darweesh, 2021-09-06:
> [image]
> 
> This picture is from an open source map named gomentum.
> Above picture shows that connection waypoints of three lanes stand at the exact same position. I believe that this is achieved by using stitch function. Am I right?
I don't know what you are talking about ! who made this map and why ! 
> 
> [image]
> 
> In my custom map, I want to do the same thing; however, the waypoints do not fit appropriately. And I am not sure that the openplanner will follow without a connection(by stitching) between them.
> 
> Lastly, I wonder is the stitch function is mandatory? Or is it just optional?
As long as the lanes are correctly connected, you don't need this function at all. and I don't recommend you to use it.

closed 2021-09-06 by hatem-darweesh