hatem-darweesh/assuremappingtools #65: Problem about traffic lights in osm format

BatuhanUlucay, 2021-12-21:
Hello Hatem,

I am working with assuremapping tool for generating custom lanelet osm maps. However, when trying to add a traffic light, I can't set its horizontal angle as needed. Is there a way to do it? Or could you explain how it calculates horizontal angle of traffic light so that I can do it manually.

Thanks.

hatem-darweesh, 2021-12-22:
Can you share a screen shot or video. 
First try to write the angle in the edit view on the right of possible,

clepz, 2021-12-24:
Hi,
I run map loader and map visualizer to use osm formatted hdmap. 
Then, I run feat_proj to check range and find a traffic light ROI from hdmap. 
Feat_proj runs without a problem and finds traffic lights in the camera frame If I use SVL-created osm hdmap. 
Feat_proj can't find the rotation of traffic lights if I use the assuremapping tool created osm hdmap. I think I created the hdmap correctly. 

The feat_proj calculates rotation from linestring points from traffic light objects. However, points are identical If I use the assuremapping tool created osm hdmap. 

I shared traffic lights related XML's both SVL-created gomentum.osm and assuremapping tool hdmap.

Gomentum:
[gomentum.txt](https://github.com/hatem-darweesh/assuremappingtools/files/7774199/gomentum.txt)

Ours:
[ours.txt](https://github.com/hatem-darweesh/assuremappingtools/files/7774211/ours.txt)