hatem-darweesh/assuremappingtools #66: Is it possible to get the source code of this package?

learningplusworking, 2022-01-04:
This work is really great and helpful.  Thank you for your great work firstly.
However, in the process of using this package to make lanelet2 map, I found that compared with the "autoware tool" on the web, it still lacks some functions. For example , traffic signs can't link to lane, can't generate lane automatically in the intersection area and no way to add sub attributes. 
Therefore, I want to do some changes on the basis of this package.But I found that this package only has lib and so files.
So I'd like to ask if I can get your source code to learn and modify. I promise I won't use it for business and I  just use it for personal learning.
If it's not appropriate, forget it. Thank you again for your great work !!!

hatem-darweesh, 2022-01-05:
Thank you for your comments. and for your offer. 
The lanelet2 development is stopped for now. one we start again maybe we can talk about an arrangement. 
my email address: hatem.darweesh@gmail.com

learningplusworking, 2022-01-05:
@hatem-darweesh okay, thanks again for your great work！

closed 2022-02-25 by hatem-darweesh