hatem-darweesh/assuremappingtools #67: How can I create assure kml file from the map.pcd?

zhongzebin, 2022-02-25:
I've read your document and watched the videos. But, I'm still quite confused about that.
First, I entered an imaginary origin point:
[image]
Then, I imported the map.pcd:
[image]
Then, I clicked auto-detection and selected the area like this:
[image]
Then, I drew the lane, like this (from the bottom view):
[image]
I selected a waypoint and checked its properties:
[image]
I can't get the lat, long and alt. I get the x, y and z. But, what it represents for? What's the unit of it?

hatem-darweesh, 2022-02-25:
1- The values you enter at the dialog are initial transformation. 
2- if you don't have GPS projection or if you are using PointCloud the GPS values will be ignored 
3- X,Y,Z are the Cartesian coordinates in meters based on the point cloud.

xfz12138, 2022-02-28:
Thx for the answer,however i am still confused that:
Which point are the X,Y,Z coordinate relative to(or where is the origin point of the point cloud)?

hatem-darweesh, 2022-02-28:
When no values entered in the Dialog box, origin is calculated at the center of the point cloud (center_x, center_y, center_z)

zhongzebin, 2022-03-04:
Thx, we've fixed that problem. But we found another problem. Can the tool support drawing a boundary like a ring (the area between a big polygon and a small polygon)? We found it can only draw boundaries with the shape of a polygon.

hatem-darweesh, 2022-03-04:
Can you explain more what do you mean by drawing a ring