hatem-darweesh/assuremappingtools #68: cannot open .osm file

lhakim85, 2022-03-03:
Hi,
I can succefully add .pcd file, but when try to load map .osm file, nothing added, and at the terminat state that "Failed to load map file: /map.osm, Error: Could not find subtype

anything that I'm missing

hatem-darweesh, 2022-03-03:
How was the map.osm file created ?

lhakim85, 2022-03-03:
I'm export it from openstreetmap.org

hatem-darweesh, 2022-03-03:
I see, we support [Lanelet2](https://github.com/fzi-forschungszentrum-informatik/Lanelet2) .osm files, not the original OpenStreetMaps.

lhakim85, 2022-03-03:
> I see, we support [Lanelet2](https://github.com/fzi-forschungszentrum-informatik/Lanelet2) .osm files, not the original OpenStreetMaps.

You mean, i need to edit/make lanelet2.osm by myself?

hatem-darweesh, 2022-03-03:
There is different tools to do so such as JOSM

lhakim85, 2022-03-03:
> There is different tools to do so such as JOSM

I'm also already try JOSM to edit .osm from openstreetmap, and save back. Still cannot import to editor.
Can you point me any resource related to JOSM,so that I can take a look.

Thanks alot @hatem-darweesh

hatem-darweesh, 2022-03-04:
kindly read the Lanelet2 documentation, it is totally different format than openstreetmaps. 
yes it is same file extension, but the XML tags are different.

lhakim85, 2022-03-04:
> kindly read the Lanelet2 documentation, it is totally different format than openstreetmaps. yes it is same file extension, but the XML tags are different.

Thanks alot @hatem-darweesh. I will read it