hatem-darweesh/assuremappingtools #71: Having issues in putting input maps origin, exporting assure.kml file into open planner and can't save the type of boundaries

SyahirMuzni, 2022-04-25:
Hi, I have some difficulties in:

1) Insert map origin and open in google earth. Seems there were 3 options, UMT, custom projection & no projection. The things that I did were: I did select custom projection, put the coordinates before uploading the .osm file - Saved it as earth.kml - Opened into google earth, but it doesn't match with the inserted coordinates. 

2) Seems like there is no option for uploading .kml in open planner.  

3) Can't save the type of boundaries. Ex: Parking, building & etc.

hatem-darweesh, 2022-04-26:
To Answer you question: 
1. can you show me screen shots and steps of what happen. 
2. you save the .kml map then use this with OpenPlanner, check op_common_params.launch and op_global_planner.launch. 
3. if you are saving OpenPlanner (ASSURE) kml type that should work, it won't work with lanelet2

SyahirMuzni, 2022-04-27:
1. Regarding the map origin:
- Selects MGRS projection, key in the coordinates.
[image]

- This is the .osm file
[image]

- saved it as 'google.kml' and open the file on google earth
[image]

- The file is successfully read, but seems like the coordinate is wrong or the projection is wrong. But I have double checked it, the coordinate is right. So I suspect I may missed some steps or key in the coordinate wrongly. 
[image]

2. Regarding OpenPlanner:
- The checkbox to open the map in .kml file is not available
[image]

hatem-darweesh, 2022-04-27:
Kindly use the projection string not the lat/lon values 
you can get the projection string you need from: 
https://spatialreference.org/
Select Proj4 , copy the string and past in the projection string box before saving the map.