hatem-darweesh/assuremappingtools #73: Vector map for Autoware

kyeops, 2022-06-06:
Hello @hatem-darweesh,

I emailed you before, and thank you for your response. I agree that talking with you here might help other people too. Basically, I would like to know if I am correct about how a vector map works for the global mapping of Autoware and discuss issues that I have with ASSURE tool.

First, I would like to create vector maps for global mapping on Autoware. I am considering vector maps that include at least information about lanes of routes that I plan to test a car on.  At the beginning of [your video](https://www.youtube.com/watch?v=5RSDDsBfxe4&t=19s), you loaded vector_map/area.csv., and the map including lane information showed up. I am wondering what map data format it is and how you created this map. I appreciate it if you can let me know any resources that describe the procedures. I am also wondering if you think I can create this without Point Cloud Data.

Plus, before I describe issues with using ASSURE tool, I am wondering if you used ASSURE tool for the vector map in the video. Right now, just from the video, I am not sure if the tool is necessary and when I really need to use the tool once I create vector maps.

Thank you for reading this.

hatem-darweesh, 2022-06-07:
To create a map you need a point of reference that represent the coordinates such as: 
1- Point cloud map 
2- .csv file contains a recorded path. 
3- Lanelet2 map. (you can create lanelet2 map using JOSM. your reference will be satellite mapping image. 

The point of reference should relate to how will you localize your robot (NDT, GNSS, etc)

kyeops, 2022-06-07:
Is it correct that I need _all of them_ that you listed to create a vector map?

Second, how did you create the path .csv? did you use Vector Map Builder from [here](https://tools.tier4.jp/)?

Thank you for your answer!

hatem-darweesh, 2022-06-08:
You can use one or more, just make sure they all have the same coordinate system and origin.

kyeops, 2022-06-09:
Currently, I am trying to avoid using pcd because of its size. Do you think I can create a map of any route based on _.csv_ without using pcd? 

Also, from [one](https://answers.ros.org/question/351978/has-anyone-been-able-to-create-working-vector-mapslanelet2-maps-for-autowareai/) of your answers, you said that it is possible to create lanelet2 once GPS is entered on ASSURE tool. Does it mean I can create a map using either JOSM or ASSURE tool? How are they different?

I am new to creating a vector map. Your answers help a lot. Thanks!